Xinyuan-LilyGO/LilyGo-HiGrow #1: AsyncTCP.h & DHT12.h libraries

erocm123, 2019-09-24:
I have followed the instructions (including copying the libaries to the appropriate place) and I am getting complaints about AsyncTCP.h & DHT12.h. Where are the proper versions of these libraries?

TTGO-Multi-function-sensor-board:5:22: error: AsyncTCP.h: No such file or directory

brunoslam, 2019-12-19:
I have the same issue with the DHT12.
Before upload the code i test the device and i can see the DHT humidity and temperature, but when i upload the sample code i can't get the values.
[image]

lewisxhe, 2019-12-20:
Sorry i forgot it, i already uploaded the used library @erocm123

erocm123, 2019-12-20:
@lewisxhe , thank you. I will try it out.

J4e8a16n, 2020-06-27:
The pins order on the probe is          
[image]
[image]

vcc-data-nc-ground.
What I see over the net for dht12 is  vcc-data-ground-nc

lewisxhe, 2020-06-28:
dth12 It supports two protocols: single bus and I2C! @J4e8a16n

closed 2020-06-28 by lewisxhe