Xinyuan-LilyGO/LilyGo-HiGrow #3: V max at 5V input ? (regulator? Schematic?)

nikil511, 2020-01-16:
@lewisxhe whats the first regulator in the 5V input?
will it accept a 6V solar panel?
(if not, would be nice to add solar capability in the "higrow lorawan shield" you are working on)

Can we have the HiGrow schematic please?

lewisxhe, 2020-01-18:
@nikil511 The schematic diagram does not seem to be provided at present. 5V is connected to VBUS. If it exceeds 5V, you need an external voltage regulator.

ro85ac, 2020-02-28:
> 
> 
> @nikil511 The schematic diagram does not seem to be provided at present. 5V is connected to VBUS. If it exceeds 5V, you need an external voltage regulator.

I am working at a project where Higrow is sending data every 2 minutes then sleeps, but the battery doesn't last more than 24hours. 

Should the battery be recharged using USB Type C plugged in T-Higrow or using a battery adapter or I am doing something wrong?

What should I do not to recharge frequently the battery? Solar panel?

jackyes, 2020-05-07:
@ro85ac 
Check this:
https://github.com/jackyes/GrowSensor
it send data every 15 min to thingspeek (customize at your need :)
The battery with that setting and a 18650 battery could run for 1 mounth+.
Yes you can charge with the onboard usb C but in my experience is too mouch slow... i prefer to take off battery once a mounth and charge it with an external charger to speed up things :)

closed 2020-06-28 by lewisxhe