Xinyuan-LilyGO/LilyGo-HiGrow #5: Schematics missing

Tobiassss, 2020-03-28:
Could you please provide the schematics of the HiGrow?

I'm trying to reduce the average current consumption to use the HiGrow battery powered.
Even when setting the ESP32 to deep Sleep and deactivating the sensor power supply (`digitalWrite(POWER_CTRL, 0);`) it draws ~8mA. This is much too much for a battery powered application. So for reducing the current when the device is basically doing nothing, I need the schematic to find the power spendthrift.

poine, 2020-05-05:
I agree, I have purchased 4 of those boards and a schematics would be nice in order to debug/improve the firmware.

pascaltippelt, 2020-06-13:
I experienced that they react very fast if you send them a whatsapp to their customer service. That way we got a schematic on github for the TTGO-OI within hours. http://www.lilygo.cn/contact.aspx?TypeId=13&FId=t7:13:7

brunoslam, 2020-06-24:
A few hours ago the schematics were uploaded https://github.com/Xinyuan-LilyGO/TTGO-HiGrow/blob/master/schematic.pdf

tyeth, 2020-06-24:
I prodded them on whatsapp as @pascaltippelt suggested, got back and uploaded within 18hrs. Serious thanks for that!

closed 2020-06-28 by lewisxhe