Xinyuan-LilyGO/LilyGo-HiGrow #8: Wifi Connection Problem

endoramax, 2020-06-14:
Hi, 

I use your ino file name is "TTGO-HiGrow". Also I have changed WIFI_SSID and WIFI_PASSWORD. I can't see any connection information and ip address in serial monitor. If I push boot button I can see smartconfig ................... endless. 

How can I connect my wifi and use this device ? 

Thank you.

lewisxhe, 2020-06-18:
[image]
[image]

endoramax, 2020-06-18:
Thank you for your support. It works after applied your instruction.

lewisxhe, 2020-06-22:
If you still have problems, please reopen.

closed 2020-06-22 by lewisxhe