Xinyuan-LilyGO/LilyGo-HiGrow #27: BAT_ADC reading on external power

asychev, 2021-03-13:
Hi!

When external power (USB-C) attached, the measured voltage on BAT_ADC (pin 33, after `(reading / 4095.0) * 2.0 * 3.3 * vRef` of course) is always around 5V, so equal to input. On the pins of battery connector it is 3.7-4.2V. Is there any way to measure battery charging voltage instead of external one during the charging phase?

Thanks!

asychev, 2021-03-18:
@lewisxhe could you please help?

asychev, 2021-04-03:
Does anybody here?..

RobFis1, 2021-04-07:
Hi!

Look into the schematics. It ist not possible to read battery votage when USB is connected. But you can determine it it's running on battery power or USB.

Kind Regards.

lewisxhe, 2021-04-12:
The hardware circuit is already fixed. When the USB is plugged in, the battery voltage cannot be read by the software.  @asychev

lewisxhe, 2021-04-12:
@RobFis1 You need to turn on and set IO4 to high level.

RobFis1, 2021-04-12:
@lewisxhe : Thank you, I already discoverd it. It's a very good power saving design. But I have trouble with AD readings after wake up from deep sleep. This seems to be a bug of the ESP32. I build a loop, if the AR readings are zero, it will trigger an ESP.restart() until the readings are OK.
Also often the I2C bus is not initializing after wakeup.

RobFis1, 2021-04-13:
I solved the ADC problem with following lines:
  adcStart(BAT_ADC);
  adcStart(SALT_PIN);
  adcStart(SOIL_PIN);
  delay(50);