Xinyuan-LilyGO/LilyGo-HiGrow #34: Cant Compile

ivanling, 2021-04-12:
I try to compile in Arduino IDE 1.8.7, 
I get the folwing error:
`
`In` file included from ...\Documents\Arduino\libraries\LilyGo-HiGrow-master\src\src.ino:6:0:

...\Documents\Arduino\libraries\ESP-DASH-master\src/ESPDash.h:38:25: fatal error: ArduinoJson.h: No such file or directory

Multiple libraries were found for "WiFi.h"
compilation terminated.

 Used: ...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi_old
exit status 1
Error compiling for board ESP32 Dev Module.`

Any Ideas?

ivanling, 2021-04-12:
If I delete Arduino wifi lib and compile I get:

In file included from C:\Users\asus\Documents\Arduino\libraries\LilyGo-HiGrow-master\src\src.ino:6:0:

C:\Users\asus\Documents\Arduino\libraries\ESP-DASH-master\src/ESPDash.h:38:25: fatal error: ArduinoJson.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.

lewisxhe, 2021-04-13:
Copy or move the folder in the lib directory to `<C:\Users\Your User Name\Documents\Arduino\libraries>` 

like :
[image]

ivanling, 2021-04-13:
Hi, Thanks a lot.
After review my files and location, is that my problem. Now working fine but I have one more issue to solve. My board has DHT11 instead of DHT12. I go try to change that in code and I think I have to use other library to.

closed 2021-04-13 by ivanling