Xinyuan-LilyGO/LilyGo-HiGrow #46: Wifi Conect Fail!

ATaciro, 2021-06-27:
Dear Sirs,
I bought 5 pieces of TTGO-T-HIGROW ESP-32 V1.1, all of which are giving the error message and not connecting to the network. My network is TP-Link Mesh DECO4 2.4Ghz and 5Ghz, with several connected devices working.
I ask for help to solve this problem.
Thank you for your attention.
ATaciro.

Message: 

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
I2C device found at address 0x23 !
done

Connect SSID:mynetwok Password:mypassword
WiFi connect fail!

thecolorblue, 2021-06-29:
Can you try connecting to a 2.4ghz only network? Are there any other networks you are able to connect to?

ro85ac, 2022-01-13:
@thecolorblue is right. You should try 2.5GHz network.

agidi, 2022-12-10:
My 2cts... most decent Wifi APs, can manage different SSIDs, i have a separate SSID on 2.4Ghz only named "IOT" .
This way when i change my main SSID password i dont have to update every gadget or TV, etc. etc. Hope this helps