Xinyuan-LilyGO/LilyGo-HiGrow #48: BME280 and MQTT

rto54, 2021-09-17:
Hello,

How can we implement the BME280, as well as an MQTT connection ?

Thanks

ro85ac, 2022-01-09:
@rto54  depends on what type of HiGrow board you have: the one with DHT12 or BME sensor?
To which server you want to make the MQTT connection ?

zialo1, 2022-02-09:
I have a similar issue. I own the BME /ESP32 sensor board and run Home Assistant.
I would like to make a connection by using mosquitto  broker if possible. Can ESP-Dash connect to the wifi and then deliver the sensor data - finally  to Home Assistant?  Just a simple one way flow of its data
Any ideas what I have to do?
Thanks

HardlineStudios, 2022-09-27:
When you first get the HiGrow device, it runs as WIFI access point and server which means the only way to view anything is to connect your wifi to it and view the web page. Kinda of annoying. 

This repository has new firmware that can connect to Home Assistant. 
https://github.com/pesor/TTGO-T-HIGrow