Xinyuan-LilyGO/LilyGo-HiGrow #51: HiGrow accuracy or missing calibration

ro85ac, 2022-01-08:
I have 4 HiGrow board and all have the same issue:
- If I put Higrow in a glass of water the **Soil Value** is always around 65%, when the expected value should be 100%
- If I take it out of the water the **Soil Value** is 22%, when the expected value should be 0%
@LilyGO @lewisxhe Is the soil sensor accuracy low or am I missing a calibration?

As a note: I made the same test with Xiaomi Soil Sensor and the values are correct.
https://www.aliexpress.com/item/32848400991.html

[image]
[image]

ro85ac, 2022-01-09:
After digging more I found that you calculate the soil moisture using method map from WMath.cpp
`long map(long x, long in_min, long in_max, long out_min, long out_max)`

And you are mapping like this: `map(soil, 0, 4095, 100, 0);`

@LilyGO  @lewisxhe Why did you send the parameters 0 and 4095 when the actual values read by the sensor are ~1475 (min) and ~3280 (max)?

LilyGO, 2022-02-21:
> After digging more I found that you calculate the soil moisture using method map from WMath.cpp `long map(long x, long in_min, long in_max, long out_min, long out_max)`
> 
> And the actual code is: `map(soil, 0, 4095, 100, 0);`
> 
> @LilyGO @lewisxhe Why did you send the parameters 0 and 4095 when the actual values read by the sensor are ~1475 (min) and ~3280 (max)?

Because the soil medium is different, it is impossible to do a unified calibration.

closed 2023-01-30 by ro85ac