Xinyuan-LilyGO/LilyGo-HiGrow #52: Soil Salt/Fertilization values units question

netmanchris, 2022-03-08:
I've got a few of these running and I've noticed that the Salt/Fertilization values level is being reported in percentage (%) rather in the typical µS/cm which I've seen in other sensors ( like the Xiaomi MiFlora for instance ) or any common plant monitor. 

I've also noticed that the percentage calculation is far above the 100% value ( some showing as high as 399% ). Can you please help me understand what the HiGrow Salt metric is actually measuring here and how are the values reported supposed to be used? 

Is the intention here to calibrate the sensors for a specific plant type and then use the percentage as the high/low mark for that specific plant type?  

Trying to better understand how to use the information that's being reported. 

thanks!

viperxeno, 2022-05-15:
Hi, I have same question... maybe is µS/cm ?

joestump, 2022-10-04:
I've done some testing with this and a `xiaomi_hhccjcy01` I have. The soil moisture sensors are identical. The lx are way, way off (LilyGo = 160s, Xiaomi = 800s under direct UV light). 

The salt/fertilization are also way different. Here's my ESPHome config:

```yaml
  - platform: adc
    pin: GPIO34
    name: "Salt"
    update_interval: 15min
    accuracy_decimals: 4    
    unit_of_measurement: 'µS/cm'
    filters:
      - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.0
          - 1.1 -> 100.0
```

In a plant that the Xiaomi measures 350 µS/cm, the LilyGO measured 98. There are, unsurprisingly, [a lot of ways to measure this](https://extension.uga.edu/publications/detail.html?number=C1019) it seems. After reading this over and searching a bit, I came across this:

[image]

Makes me wonder if this is returning `EC(dS-1)`? I dunno, but would like to get this reading similar results to the Xiami's if possible.