Xinyuan-LilyGO/LilyGo-HiGrow #60: two improvements

edegraaff, 2022-10-30:
Gettign msqtt and lora the thingsnetwork would be great then you can add historycal data and dashboards to the project

hekopath, 2022-10-31:
Lora is already availabel as a basic example in this project for the Lora module:
[https://github.com/Xinyuan-LilyGO/LilyGO-T-HiGrow-LoRaWLAN](https://github.com/Xinyuan-LilyGO/LilyGO-T-HiGrow-LoRaWLAN)

edegraaff, 2022-11-04:
Thanks, do you think there will be some updates in the code, sice i see some major issues at the moment one is the quick battery drain and some mesurements of the sensors seems to be not ok.

hekopath, 2022-11-04:
No, i do not expect updates on the LoraWAN repo.
In this issue is described how you can enable deepsleep: [https://github.com/Xinyuan-LilyGO/LilyGO-T-HiGrow-LoRaWLAN/issues/5](https://github.com/Xinyuan-LilyGO/LilyGO-T-HiGrow-LoRaWLAN/issues/5)