Xinyuan-LilyGO/LilyGo-HiGrow #61: Strange readings in my LilyGrow

edegraaff, 2022-10-30:
Hi, i flashed my lilygo but the reasings i get are quite strange. Do i have the wrong sensors defined or something else not ok? 

[image]

LilyGO TTGO T-Higrow ESP32 - BME280 Sensor

Any clue?

KamilSzczepankiewicz, 2022-12-13:
I believe issue is with wrong calculation of Humidity, as I mentioned in the issue above yours.

edegraaff, 2022-12-13:
> I believe issue is with wrong calculation of Humidity, as I mentioned in the issue above yours - at least for humidity readings

And the Saltvalues? it seems the sensors are delivering time over time differend readings, and or incorrect ones. When the sensors are dry, moist and salt need to be 0 since it is dry there can't be any conduct values other than 0 right?

Amr-MKamal, 2023-06-20:
Quick solve : just divide the value by 10 you will get soil mix ec at ds/cm  ( based on TCR555 configuration )

edegraaff, 2023-06-20:
Would be nice if someone checks the code and make an update, would be nice if i get the whole working in the propper way. Also with enough battery life time