Xinyuan-LilyGO/LilyGo-HiGrow #67: Compilation problems with LILYGO® T-Higrow ESP32 Soil Tester DHT11 BEM280 Kit.

Copyrighter, 2024-06-05:
Hi,

I'm just received LILYGO® T-Higrow ESP32 Soil Tester DHT11 BEM280 Kit and trying to follow instructions and connect it to WIFI.

But, I get two errors during compilation:
1. Compilation error: 'GPIO_SEL_35' was not declared in this scope; did you mean 'GPIO_NUM_35'?
```
C:\Users\Any1\Documents\Arduino\src\src.ino: In function 'void sleepHandler(Button2&)':
C:\Users\Any1\Documents\Arduino\src\src.ino:119:34: error: 'GPIO_SEL_35' was not declared in this scope; did you mean 'GPIO_NUM_35'?
  119 |     esp_sleep_enable_ext1_wakeup(GPIO_SEL_35, ESP_EXT1_WAKEUP_ALL_LOW);
      |                                  ^~~~~~~~~~~
      |                                  GPIO_NUM_35

exit status 1
```

2. Compilation error: text section exceeds available space in board
```
Sketch uses 1413085 bytes (107%) of program storage space. Maximum is 1310720 bytes.
Global variables use 50396 bytes (15%) of dynamic memory, leaving 277284 bytes for local variables. Maximum is 327680 bytes.
```

Who can help?

Thank You.

edegraaff, 2025-01-21:
I have the same problem, i found some documentation where is stated you need to downgrade the esp32 boards to a lower version.

edegraaff, 2025-01-24:
Downgrading to esp boards 2.08 delivers a different error it seems still a declare arror.

\Arduino\HiGrow\src\src.ino:118:34: error: 'GPIO_SEL_' was not declared in this scope

                                  ^~~~~~~~~
o\HiGrow\src\src.ino:118:34: note: suggested alternative: 'GPIO_SEL_0'
     esp_sleep_enable_ext1_wakeup(GPIO_SEL_,35 ESP_EXT1_WAKEUP_ALL_LOW);
                                  ^~~~~~~~~
                                  GPIO_SEL_0
iGrow\src\src.ino:118:46: error: expected ')' before 'ESP_EXT1_WAKEUP_ALL_LOW'
     esp_sleep_enable_ext1_wakeup(GPIO_SEL_,35 ESP_EXT1_WAKEUP_ALL_LOW);
                                 ~            ^~~~~~~~~~~~~~~~~~~~~~~~
                                              )